gdsports/USB_Host_Library_SAMD #2: USB MIDI Host with HUB Problems

DatanoiseTV, 2019-01-16:
I'm working on USB MIDI Host support with a USB hub and I cannot get the device to enumerate properly. I have ported changes from the USB Host Shield Library 2.0.

Here are my changes:
https://github.com/DatanoiseTV/USB_Host_Library_SAMD

Any hints?

gdsports, 2019-01-23:
I just verified the USB hub fixes from https://github.com/felis/USB_Host_Shield_2.0/pull/414 work on the USB Host Shield. I will see why they do not work on SAMD.

billmoser, 2020-07-03:
I had this issue also.  The in particular, the example in USBH_MIDI/USB_MIDI_converter_multi only worked for me if I had a single controller on the hub.  I used @DatanoiseTV's fork, and everything works ok reading midi from two controllers.  For a project I'm working on (not yet in github) I need to be able to write the devices as well.  Here things did not go so well, even with @DatanoiseTV's changes, until I refactored the example code to wait in setup() for the devices to be ready before going into reads/writes in loop().  Even so,  it takes about 45 seconds for my devices (Source Audio guitar effects pedals) to be ready to go into the read/write loop, which seems excessive.  I forked your repo, and then merged in @DatanoiseTV's changes, so my fork is current with your repo.  Let me know if you want me to do a pull request.  BTW, Thanks for doing this port of @felis's UHS2! My goal was to be able to debug/test an Arduino Uno using UHS2, and deploy with your SAMD port on my Trinket M0, and that  looks like it's going to work out at this point.