gdsports/USB_Host_Library_SAMD #4: Problem with USBH_MIDI with Zoom G3.

soulwyx, 2019-08-22:
I can make USB_STATE_RUNNING with my Zoom G3 without de USBH_MIDI library.
But when I add the:
USBH_MIDI midiUsb(&usb);
I get problems.
Stuck a while with USB_STATE_CONFIGURING and then USB_STATE_ERROR.
Any Idea of what is happening or what I need to look for?

I just need to send data. Another library that I can use to send data?
outTransfer(bAddress, epInfo[epDataOutIndex].epAddr, bytes_send, dataptr);
How I get the "bAdress" and "epInfo[epDataOutIndex].epAddr" without USBH_MIDI library?

gdsports, 2019-08-22:
Post your code and any serial console debug output.

soulwyx, 2019-08-22:
`#include <Arduino.h>
#include <usbhub.h>

USBHost usb;

void setup() {
	Serial.begin(9600);
	usb.Init();
}

void loop() {
	usb.Task();
	delay(1000);
}`

Code that works.
Debug output:
 + USB_DETACHED_SUBSTATE_INITIALIZE

 + USB_ATTACHED_SUBSTATE_RESET_DEVICE

 + USB_ATTACHED_SUBSTATE_WAIT_RESET_COMPLETE

 + USB_ATTACHED_SUBSTATE_WAIT_SOF

 + USB_ATTACHED_SUBSTATE_WAIT_SOF

 + USB_STATE_RUNNING

`#include <Arduino.h>
#include <usbhub.h>
#include <usbh_midi.h>

USBHost usb;
USBH_MIDI midiUsb(&usb);
void setup() {
	Serial.begin(9600);
	usb.Init();
}

void loop() {
	usb.Task();
	delay(1000);
}`

Code that doesn't work.
Debug output:
+ USB_ATTACHED_SUBSTATE_RESET_DEVICE

 + USB_ATTACHED_SUBSTATE_WAIT_RESET_COMPLETE

 + USB_ATTACHED_SUBSTATE_WAIT_SOF

 + USB_STATE_CONFIGURING

/!\ USBHost::Task : USB_STATE_CONFIGURING failed with code: 8

soulwyx, 2019-08-22:
Look likes the problem is on AttemptConfig function. 
Any idea?

gdsports, 2019-08-23:
What SAMD board are you using? If you are using an Uno, Mega, or Leonardo with a USB host shield, look in the https://github.com/felis/USB_Host_Shield_2.0.

I do not see how sketch 1 could work. Most SAMD boards have only 1 USB port which this library switches to USB host mode so serial console debug ouput does not come out on the same port. Serial console debug output must be sent to Serial1 or other Serial port.

By the way, remove the delay(1000) because it is not needed.

If the Zoom G3 is not MIDI class compliant it will not work.

There is one program in github that indicates the device is usable with the old USB Host Shield library so it may be MIDI class compliant. This uses the older USB host shield so does not work with this library.

[https://github.com/vegos/ZoomG3_ArduinoMIDI/blob/master/ZoomG3_ArduinoMIDI.ino](https://github.com/vegos/ZoomG3_ArduinoMIDI/blob/master/ZoomG3_ArduinoMIDI.ino)

soulwyx, 2019-08-23:
I'm using arduino zero. So I can serial debug and use the USB Host application.
A lot of people make works the Zoom G3 with USB Host Shield. 
Here randomly work. I think the problem is on AttemptConfig. For a unknown reason, the code stucks in there. Anything that I can try to change?

soulwyx, 2019-08-23:
I don't know why, now it's working.  Maybe a problem in my Zoom G3, or noise problem. I'm feeling stupid.

gdsports, 2019-08-23:
The Zero is a good board for USB host debugging since it has two USB ports. I am glad to hear it is working.