gdsports/USB_Host_Library_SAMD #13: Recvbuf needs to be aligned

studiohsoftware, 2020-12-03:
Hi the datasheet for the SAMD21 here https://cdn.sparkfun.com/datasheets/Dev/Arduino/Boards/Atmel-42181-SAM-D21_Datasheet.pdf page 836, says that the ADDR value must be aligned with the two least significant bits zero.

So in USBH_MIDI_dump.ino the bufMidi declaration should be
uint8_t bufMidi[64] __attribute__ ((aligned (4)));

Similarly, looking at USB_MIDI_converter_multi.ino, this leads to the declaration in usbh_midi.h 
uint8_t recvBuf[MIDI_EVENT_PACKET_SIZE];
This also needs to be aligned per the datasheet.

I am not sure the implications of this. I am doing a deep dive on this code to sort out the dropped MIDI message problem, and just stumbled on it.

todbot, 2020-12-06:
I was really hopeful your hypothesis that the alignment of`bufMidi` and `recvBuf` would address the message loss I'm seeing (described in #8).  But I dusted off my test setup and tried it out: nope. Still seeing missing MIDI messages.
But this is the kind of issue  I'm suspecting is the problem.

studiohsoftware, 2020-12-06:
yeah I know. seems to make no difference. I am just about done with an
interrupt based multi midi ino example.

I did find as well that interval set to zero throws reset errors but while
using a 1 there in the interrupt case fixes it, there is no benefit to the
polling approach used in the examples.

Short version is that the polling approach is just horrible.