gdsports/USB_Host_Library_SAMD #14: USB Not starting on Adafruit QTPy w/USB-C->USB-A (3.0) Adapter

binko, 2020-12-10:
Hi folks,

I made an adapter PCB that puts two Adafruit QTPy boards back-to back and connects the UART, SPI and I2C lines (plus power and ground).  Happy to share the Eagle files when it's working (or before privately).  

However, when I run the MouseController example it only gets "Detached, waiting for Device" and never enumerates.  The USB_desc example also just waits with no state changes.

Because the QTPy has only a USB-C connector, I am using a USB-C OTG connector (this one: https://tinyurl.com/y6rd252y )

Is there any chance that the USBHost code won't work w/a USB-C or a USB 3.0 connection?  

This same code worked with the Trinket M0, so I'm a bit lost.

Thanks

ATMakersBill, 2020-12-15:
Just an update, I have tried every conceivable combination of USB-C device & adapter including the adapter above, others that were not listed as USB 3.0, those that specifically said OTG as well as a direct USB-C Mouse (with no adapter).

The QT Py never gets beyond "Detached, waiting for device" and nothing ever enumerates.

Any guidance would be appreciated.

ATMakersBill, 2021-06-10:
Need to update and close this issue - the problem is the protection diode on that board.  In the new QT Py 2040, there is a jumper to short it - but with it in place, which is true on the M0 QT Py, the board will not power the USB device.

So, the QT Py M0 can't be used as a USB Host without removing that diode and shorting the connection.

cbmainz, 2022-02-15:
Good to know, I was going to use two QT Py actually for this reason. Then I stick with two Trinkets M0.