gdsports/USB_Host_Library_SAMD #22: SparkFun SAMD51 MicroMod Not Showing Up on the COM Port After Example Sketch

Kuantronic, 2021-12-11:
I'm using the SAMD51 MicroMod with the ATP Carrier board. When the default MouseController example sketch is uploaded, the board is no longer detected by my PC. Is there a way to get it show on the COM port so I can upload new sketches??

Kuantronic, 2021-12-12:
I figured out the COM port problem. Just needed to double press the reset button to set it in bootloader mode. But I still can't seem to get the example code to display my mouse inputs to the SAMD51.

sdetweil, 2022-02-06:
well, the debugging messages come out on Serial1  (rx/tx), so you have to have something connected there.
I have another Nano with rx/tx cross connected to the nano being the USB host device. 
the reporting nano is running the receiver sketch from the Serial examples
https://docs.arduino.cc/tutorials/nano-33-iot/UART

Serial (the usb port) , is in host mode, waiting for a client device to connect...  the PC is ALSO in host mode.

closed 2022-02-07 by Kuantronic